revtel/react-native-nfc-manager #715: Can't retrieve tag information with background tag on IOS

ImperialCrowns, 2024-04-18:
Hey !

I wrote the following code :

```typescript
const initNFC = () => {
  const onBackgroundTag = async (tag: TagEvent | null) => {
    console.log('onBackgroundTag', tag);
  };

  const initalizeBackgroundTag = async () => {
    // changed
    await NfcManager.start();

    const bgTag = await NfcManager.getBackgroundTag();
    onBackgroundTag(bgTag);

    // listen to other background tags after the app launched
    NfcManager.setEventListener(NfcEvents.DiscoverBackgroundTag, onBackgroundTag);
  };
  initalizeBackgroundTag();
};

function App() {
  initNFC();
  return (
    <View>
      <Text>Hello World</Text>
    </View>
  );
}

export default App;
```

I'm trying to retrieve tag informations when reading tag in background but the return value of NfcManager.getBackgroundTag is always null.
What am I doing wrong ? Is there something I miss ? Is it possible to do it ?

### Environment:
- Device: Iphone 14 Pro Max
- iOS: 17.4.1
- react-native-nfc-manager@3.14.14
- Universal Links implemented and fully working

I've already tried many solutions proposed in previous issues, and it don't work.

BhavyaCodeAlchemy, 2024-04-19:
It's Working fine in android ?

ImperialCrowns, 2024-04-23:
Yes, on Android it works perfectly.

gigeos, 2024-04-23:
Same for me

I can't retrieve the tagId when the application is launched in background on iOS thanks to the little NFC notification that appears when a tag is touched.

closed 2024-08-07 by github-actions[bot]

wuguishifu, 2024-10-09:
I'm also having this issue, I can't get any scan information.

Also, if the issue is tagged for iOS, saying that it works perfectly on Android is probably not going to be relevant information.