revtel/react-native-nfc-manager #721: Session sometimes closes itself

pke, 2024-05-09:
I recon on iOS (Android not tested) an open scan session sometimes closes itself. This means the iOS bottom sheet that displays the alert messages is closed and there is no more active session. 
Is this a default timeout somewhere?

giuliobracci, 2024-06-10:
Hi @pke, as I recon, there's no way on iOS to set that timeout, but the timeout is treated as an NfcError, so you can check against it when you receive an error. By the way, I've not used the library with its listeners but with only more imperative solutions due the kind of app I'm developing, but I think that you can listen for NfcErrors.

Here's an example of how I've implemented it. 

```
try {
  // Modal should appear
  await NfcManager.requestTechnology(NfcTech.Ndef, {
    alertMessage: t('holdYourPhone')
  });
} catch (err) {
  if (err instanceof NfcError.Timeout) {
  // Here's the timeout.
  }
} finally {
 // do stuff.
}
```

There's also a `NfcError.UserCancel` whitch can come very handy to know when the user has pressed on the cancel button.

pke, 2024-06-13:
@giuliobracci have you tested this? This causes a "uncaught exception" in my app when the timeout occurs.

closed 2024-09-27 by github-actions[bot]