revtel/react-native-nfc-manager #740: Library is not working on iOS while working well on Android

17awakil, 2024-07-23:
I'm not even able to scan an NFC tag on iOS, while the same code works well on Android.

Is anyone else experiencing this?

How have you addressed this?

Thanks for your help!

Jidahan, 2024-07-29:
i have same problem! sos!
[image]

tastydev, 2024-08-11:
Heys Guys, you are probably calling `NfcManager.start()` multiple times make sure to call it only once in your lifecycle.

aberikashvili, 2024-10-17:
I have the same problem, NFC popup opens on iPhone, but it doesn't detect anything

subhusal, 2024-10-22:
For iOS it seems, if the ndefMessage is null, it won't work. But for the same card it works fine for Android.

Can anyone have a way out for this issue?

closed 2025-02-05 by github-actions[bot]

Youngermaster, 2025-03-09:
PLEASE RESPONDDDD!!!!! WE NEED THIS ASAP. PLEAS TEAM HELP US