revtel/react-native-nfc-manager #745: When scanning, AppState  becomes background

valery-lavrik, 2024-08-26:
I have listening enabled

```
        AppState.addEventListener('change', (state) => {
                console.log('LOGAUT', state);
        });
```

When the scan event is triggered
```
    NfcManager
        .requestTechnology(NfcTech.Ndef);
```

Apostate becomes = 'background'

Is there any way to fix this?

pke, 2024-09-03:
I guess you are on iOS and the answer is no. The OS puts your app in the background and its own scan dialog in the foreground.
This type of question would be better suited for the [Disscussions Q&A format](https://github.com/revtel/react-native-nfc-manager/discussions)

valery-lavrik, 2024-09-04:
No, I use android

pke, 2024-09-04:
Thats unexpected, since Android does not put up its own NFC dialogue in front of your app.

valery-lavrik, 2024-09-05:
> Thats unexpected, since Android does not put up its own NFC dialogue in front of your app.

But that's exactly what happens

sweebee, 2024-09-19:
I've got the same issue (on android), its very annoying since i'm using the android presentation api for an external display, but when the app goes to background, the display goes away for a few miliseconds.

closed 2025-01-04 by github-actions[bot]

przemoch-dev, 2025-01-20:
Bump. Same here

przemoch-dev, 2025-01-20:
Bump. Same here