revtel/react-native-nfc-manager #747: NDEF scanner app works on  IOS but not on Android

kitkline, 2024-09-19:
Hi all, advice or help appreciated (ps - awesome tool!)

Following the documentation we have built NFC tag scanning into our app and works perfectly on IOS but on Android the tag is not read (to tag is ever returned from await NfcManager.getTag()

Checked - the phone is NFC capable and the app NFC tools on the phone can read the tag (i.e issue is in our code  - not in the phone).

Checked - AndroidManifest.xml contains:  <uses-permission android:name="android.permission.NFC" />

After the tech is launched by the user tapping the 'scan' button there is no error but scanning the tag does nothing. Pairing the phone with Android studio we see the following in logcat:

2024-09-19 23:05:01.901 14222-14419 ReactNativeNfcManager   org.gearlog.app                      D  isSessionAvailable: false
2024-09-19 23:05:01.905 14222-14419 ReactNativeNfcManager   org.gearlog.app                      D  registerTagEvent
2024-09-19 23:05:01.908 14222-14419 ReactNativeNfcManager   org.gearlog.app                      I  enableForegroundDispatch, enable = true

And then nothing else

Phone is Samsung A10 running Android 11 (it's a cheap old phone we used for physical testing - but should work given that NFC tools works?)

How can we generate more debug or logging to help resolve?

Code follows...
```

  async scanNFC() {
    try {
      alert('We see this message');
      await NfcManager.requestTechnology(NfcTech.Ndef);
      alert('We never see this message');
      const tag = await NfcManager.getTag();

       -> our tag parsing code follows but we never get to that as never get a tag

     
    } catch (ex) {
     	-> our error alerting
     	
      );
    } finally {
        NfcManager.cancelTechnologyRequest();
    }
  }

```

kitkline, 2024-09-20:
To add further to this we have boiled the code down to the absolute minimum and the basic issue (Android) only is that 'await NfcManager.getTag()' simply never resolves. The NFCManager is running and once the app is on that screen the normal android phone default response to a nearby tag no longer occurs - so clearly the app has grabbed NFC as the manager starts - but when an tag is close NfcManager.getTag() does not resolve. Could this be something to do with how android sends the tag event on older phones (A10 running Android 11 which can run the app NFC tools just fine)

kitkline, 2024-09-23:
To add further to this we have also tried on a Google Pixel 5 running Android 14 with the exact same result. In summary the most basic minimal example of using this library works perfectly on an iOS device but does not detect the same tag either on Android 10 or Android 14 with the issue being that that 'await NfcManager.getTag()' simply never resolves. This leaves us confused as this cannot be the case and there must be many users developing against Android with this device and yet we cannot get a couple of code lines of minimal example to work? Any suggestions much appreciated :)

kitkline, 2024-09-23:
To futher add to this we have also tried the below method of using NFCmanager on Adroid with the exact same result - the await for the tag never resolves....  

   ```
       NfcManager.setEventListener(NfcEvents.DiscoverTag, tag => {
            alert("tag found");
             
          });
          await NfcManager.registerTagEvent();
```

javix64, 2024-11-14:
hey @kitkline which kind of card are you using?
what is the output of reading card in iPhone?
is there any information about the card that you are trying to read?

closed 2025-02-28 by github-actions[bot]