revtel/react-native-nfc-manager #748: Android 14 (API level 34) Support - cannot find symbol nfcAdapter.setNdefPushMessage(msgToPush, currentActivity);

marconioliveira, 2024-09-19:
Hi everyone, Google Play Store is blocking updates for apps that are built below  34 API level. When changing the build level to 34, I'm getting the following build error:

`/$APP_PATH/node_modules/react-native-nfc-manager/android/src/main/java/community/revteltech/nfc/NfcManager.java:967: error: cannot find symbol

                    nfcAdapter.setNdefPushMessage(msgToPush, currentActivity);
                              ^
  symbol:   method setNdefPushMessage(NdefMessage,Activity)
  location: variable nfcAdapter of type NfcAdapter
Note: /$APP_PATH/node_modules/react-native-nfc-manager/android/src/main/java/community/revteltech/nfc/NfcManager.java uses or overrides a deprecated API.`

NfcAdapter changes: https://developer.android.com/sdk/api_diff/34/changes/android.nfc.NfcAdapter`

I don't have experience in native code, but if I find a solution, I'll post it here.

Thanks

BenHurMartins, 2024-12-03:
Hi, any update, facing same issue.

jn222, 2024-12-10:
Same here! Any updates @marconioliveira ?

marconioliveira, 2024-12-10:
@jn222 I solved this by creating a new project with the latest version of react native and importing the most current versions of the libraries I used. I was trying to avoid that, but it was good for the project.

My project dependencies: 

{
  "name": "xyz",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "lint": "eslint .",
    "start": "react-native start",
    "test": "jest",
    "postinstall": "patch-package"
  },
  "dependencies": {
    "@bam.tech/react-native-image-resizer": "^3.0.10",
    "@flyerhq/react-native-android-uri-path": "^2.3.0",
    "@likashefqet/react-native-image-zoom": "^4.1.0",
    "@notifee/react-native": "^9.0.2",
    "@react-native-community/slider": "^4.5.3",
    "@react-native-firebase/analytics": "^20.5.0",
    "@react-native-firebase/app": "^20.5.0",
    "@react-native-firebase/auth": "^20.5.0",
    "@react-native-firebase/dynamic-links": "^20.5.0",
    "@react-navigation/drawer": "^6.7.2",
    "@react-navigation/native": "^6.1.18",
    "@react-navigation/native-stack": "^6.11.0",
    "cheerio": "1.0.0-rc.12",
    "fbemitter": "^3.0.0",
    "formik": "^2.4.6",
    "lottie-react-native": "^7.0.0",
    "moment": "^2.30.1",
    "patch-package": "^6.5.1",
    "react": "18.3.1",
    "react-native": "0.75.3",
    "react-native-animated-pagination-dots": "^0.1.73",
    "react-native-audio-recorder-player": "^3.6.11",
    "react-native-background-upload": "github:daraujo-liberson/react-native-background-upload#v7.2.7",
    "react-native-blob-util": "0.19.6",
    "react-native-compressor": "^1.9.0",
    "react-native-document-picker": "^9.3.1",
    "react-native-draggable-flatlist": "^4.0.1",
    "react-native-fast-image": "^8.6.3",
    "react-native-file-viewer": "^2.1.5",
    "react-native-fs": "^2.20.0",
    "react-native-gesture-handler": "^2.20.0",
    "react-native-get-random-values": "^1.11.0",
    "react-native-image-crop-picker": "^0.41.2",
    "react-native-image-picker": "7.1.2",
    "react-native-keyboard-aware-scroll-view": "^0.9.5",
    "react-native-linear-gradient": "^2.8.3",
    "react-native-mmkv": "2.10.1",
    "react-native-modal": "^13.0.1",
    "react-native-nfc-manager": "3.16.0",
    "react-native-orientation-locker": "^1.7.0",
    "react-native-pager-view": "^6.4.1",
    "react-native-paper": "^5.12.5",
    "react-native-pdf": "^6.7.5",
    "react-native-qrcode-svg": "6.3.1",
    "react-native-reanimated": "^3.15.3",
    "react-native-reanimated-carousel": "^3.5.1",
    "react-native-receive-sharing-intent": "^2.0.0",
    "react-native-safe-area-context": "^4.11.0",
    "react-native-screens": "^3.34.0",
    "react-native-share": "^11.0.3",
    "react-native-sqlite-storage": "^6.0.1",
    "react-native-svg": "14.1.0",
    "react-native-svg-transformer": "^1.5.0",
    "react-native-tab-view": "3.5.2",
    "react-native-track-player": "^4.1.1",
    "react-native-url-polyfill": "^2.0.0",
    "react-native-vector-icons": "^10.2.0",
    "react-native-video": "^6.6.2",
    "react-native-vision-camera": "4.5.2",
    "tinycolor2": "^1.6.0",
    "uuid": "^9.0.0",
    "yup": "^1.4.0"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@babel/preset-env": "^7.20.0",
    "@babel/runtime": "^7.20.0",
    "@react-native/babel-preset": "0.75.3",
    "@react-native/eslint-config": "0.75.3",
    "@react-native/metro-config": "0.75.3",
    "@react-native/typescript-config": "0.75.3",
    "@types/react": "^18.2.6",
    "@types/react-native-vector-icons": "^6.4.18",
    "@types/react-test-renderer": "^18.0.0",
    "@types/uuid": "^10.0.0",
    "babel-jest": "^29.6.3",
    "eslint": "^8.19.0",
    "jest": "^29.6.3",
    "prettier": "2.8.8",
    "react-test-renderer": "18.3.1",
    "typescript": "5.0.4"
  },
  "engines": {
    "node": ">=18"
  }
}

closed 2025-03-26 by github-actions[bot]