revtel/react-native-nfc-manager #753: android: property "compileSdkVersion" cannot be found or does not have a numeric value.

giovanni-caiazzo, 2024-10-16:
Hi! I'm using the plugin for `react-native-nfc-manager` and this is my plugin config:
```json
[
     "react-native-nfc-manager",
     {
          "nfcPermission": "The app needs to be able to read NFC tags so that it can start or end employees' turns",
          "includeNdefEntitlement": false
     }
],
```
If I'm using the plugin, when running `npx expo prebuild` I get the warning
`android: withBuildScriptExtVersion: Cannot set minimum buildscript.ext.compileSdkVersion version because the property "compileSdkVersion" cannot be found or does not have a numeric value.`.

If I remove the plugin, the warning is gone... However I need the plugin.

Is it possible to fix this annoying warning? It doesn't seem like it is making something break, but it would be nice to fix it.

chriswheeler, 2025-02-14:
We are also seeing this warning with Expo 52.0.33 react-native-nfc-manager 3.16 and "react-native-nfc-manager" added to 'plugins'

superyarik, 2025-09-01:
We are also seeing this warning with Expo 53.0.22 react-native-nfc-manager 3.16.3 and "react-native-nfc-manager" added to 'plugins'

trentrand, 2025-09-20:
@whitedogg13 can you reopen this issue?