revtel/react-native-nfc-manager #757: upgrade @expo/config-plugins dependency to 8 for support with New Architecture

jaimemarijke, 2024-11-13:
I'm following instructions from the https://expo.dev/changelog/2024/11-12-sdk-52#testing-your-app-with-the-new-architecture and am blocked by an old dependency of `react-native-nfc-manager`.

'''
$ npx expo-doctor@latest      
Enabled experimental React Native Directory checks. Unset the EXPO_DOCTOR_ENABLE_DIRECTORY_CHECK environment variable to disable this check.
✔ Check package.json for common issues
✔ Check Expo config for common issues
✔ Check native tooling versions
✔ Check if the project meets version requirements for submission to app stores
✔ Check for common project setup issues
✔ Check dependencies for packages that should not be installed directly
✔ Check for app config fields that may not be synced in a non-CNG project
✔ Check npm/ yarn versions
✔ Check for issues with Metro config
✔ Check for legacy global CLI installed locally
✔ Check that native modules do not use incompatible support packages
✖ Validate packages against React Native Directory package metadata
✔ Check Expo config (app.json/ app.config.js) schema
✔ Check that packages match versions required by installed Expo SDK
✖ Check that native modules use compatible support package versions for installed Expo SDK

Detailed check results:

Expected package @expo/config-plugins@~9.0.0
Found invalid:
  @expo/config-plugins@8.0.8
  (for more info, run: npm why @expo/config-plugins)
Advice: Upgrade dependencies that are using the invalid package versions.

$ $ npm why @expo/config-plugins@8.0.8                                                                                                                                                                                                                                                                             1 ↵
@expo/config-plugins@8.0.8
node_modules/@expo/config-plugins
  @expo/config-plugins@"~8.0.0" from react-native-nfc-manager@3.16.0
  node_modules/react-native-nfc-manager
'''

Patrichlidin, 2024-11-16:
This may not be a long term fix, but i did it like this just to get it work:
package.json:

```
  "overrides": {
    "react-native-nfc-manager": {
      "@expo/config-plugins": "^9.0.9"
    }
  },
```

Its works for me, have only tried with dev-build for android atm.

trajano, 2024-11-16:
@AlexandraOlegovna please don't set it to 9.0.9 and up instead set it to '8 || 9' as some of us may still be on older versions of Expo.

RonakDoshiTMI, 2024-11-20:
Fixed by #758

closed 2025-03-06 by github-actions[bot]