revtel/react-native-nfc-manager #763: Host Card Emulation

jlkalberer, 2025-01-03:
Hi,

I am doing a bit of work to add Host Card Emulation since [iOS now supports it as of 17.4.](https://developer.apple.com/documentation/corenfc/cardsession)

I was wondering how you'd prefer I expose these APIs (or if you even want them in this package)?
Should I try to shoe-horn it into `registerTagEvent` or just add a new API specific to this feature?

I am tackling Android first since I'm more comfortable there but I just wanted to get some feedback before I open a PR.

whitedogg13, 2025-01-05:
Hi John,

That will be great!

I would love to see this feature included in current package. The main
reason is:
- When users develop a HCE based app, they normally need a corresponding
reader side app. A library integrate both sides will make the
developement easier.

For the API side, I will suggest to separate these new APIs from any
existing APIs, including "registerTagEvent", because:
- API naming: for HCE, we don't listen to "tag", but wait for the "reader"
- flow: card emulation normally requires more setup flow than tag reading

Look forwards to see your upcoming PR, and I will definitely look into this
with you!

Thanks

Richie

DylanWard14, 2025-02-26:
Hi!

Any update on this?

Thanks

jkalberer, 2025-02-26:
Life got in the way a bit haha. 

I'll try to push up the changes I made. I can't remember where I left off with things but I think I broke my NFC chip as I was working on it.

DylanWard14, 2025-02-26:
No worries. Would love to see this feature implemented.

bitsmyth, 2025-02-27:
That would be great! 

+1

jlkalberer, 2025-03-10:
I won't be able to work on this further but it's pretty close to being done (I think). https://github.com/revtel/react-native-nfc-manager/pull/770

I tried to set things up so that it works with existing codepaths instead of having a bunch of net-new code.

Also, I was only focused on having the card emulate a tag + data. I wasn't making it programmable or anything.

closed 2025-06-25 by github-actions[bot]

icedevml, 2025-09-13:
Just a note, I've implemented a separate module that covers the HCE for both Android and iOS: https://github.com/icedevml/react-native-host-card-emulation

It might be a helpful reference if you would like to port that stuff into React Native NFC Manager, otherwise I welcome any pull requests as well.