revtel/react-native-nfc-manager #774: Reading multiple non-NDEF tags in a single session - iOS

henrycowan, 2025-04-11:
I've followed along with your newline vids and the `NfcManager.registerTagEvent() ` works well for scanning multiple NDEF formatted cards in a persisting NFC session. The session opens, we tap away, and everything works well. 

I now need to implement a similar functionality for non-NDEF formatted tags. I've attempted using `NfcManager.requestTechnology()` and polling with `NfcManager.getTag()`, but this function seems to return a previously cached tag with no way of flushing the previous tag info. The only solution seems to be to requestTech, scan tag, cancel request, re-request, scan tag and so on, but this causes the native NFC dialog to open and close (very slowly). I understand there are iOS imposed cool-off period limitations for restarting NFC sessions. 

Is there any way to read multiple tags, non-NDEF formatted, in quick succession, without opening and closing the NFC session per-tag? Basically, replicate what you have achieved with the Tag Counter game but with non-NDEF formatted cards.

Seems I'm not the only one after this functionality - https://github.com/revtel/react-native-nfc-manager/issues/713 

Thanks in advance!

henrycowan, 2025-04-29:
Hi are there any updates on the above issue?

dcanora, 2025-05-19:
Have you gotting this working? I'm successfully using `NfcManager.restartTechnologyRequestIOS()` instead of `NfcManager.requestTechnology()` for reads after the first one on iOS.  

Calling `NfcManager.requestTechnology()` repeatedly works fine for me on Android.

arahman69, 2025-05-21:
works fine for me, thx @dcanora

closed 2025-09-05 by github-actions[bot]